MapIV/eagleye #35: Sample data

lijecaru, 2020-05-26:
How to use the sample data to test? I am playing the rosbag file and check that the topics `/imu/data_raw` and `/can_twist` are publishing. Then I start RTKLIB with `bash rtklib_ros_bridge_sample.sh` I get the following error when changing the path to /dev/ttyACM0:

```
no navigation data: rtkrcv.nav

** rtkrcv ver.2.4.3 b31 console (h:help) **
rtkrcv> rtk server start error (str1 open error path=/dev/ttyACM0)
```
If I leave the original path, then I get a server start error:

`rtkrcv> rtk server start error (str1 open error path=/serial/by-id/usb-u-blox_AG_-_www.u-blox.com_u-blox_GNSS_receiver-if00:9600:8:n:1:off)`

OsamuSekino, 2020-05-28:
Thank you for your interest.
The sample data already contains /rtklib_nav, so you don't need to start RTKLIB.
Therefore, there is no need to prepare an actual machine.

The way to try the sample data is simple.
```
rosbag play --clock eagleye_sample.bag
roslaunch eagleye_core eagleye_localization.launch
```
The estimated results will be output about 100 seconds after playing the rosbag. This is because we need to wait for the data to accumulate for estimation.

lijecaru, 2020-05-29:
It works well, thank you!

closed 2020-05-29 by lijecaru