MapIV/eagleye #37: Is it possible to localization without using actual machine and RTKLIB?

mcomic, 2020-06-26:
I have IMU and GNSS data of a traffic scene, and I'm trying to use ROS nodes to send data directly to eagleye_core and to fetch the output.

Judging by the sample data, eagleye seems able to do the localization without RTKLIB and actual machine.

So I wondering is it possible to generate all the necessary information that eagleye needs to localization without using RTKLIB and actual machine?
If yes, could you please also tell me which topics should be published from outside to make eagleye perform properly?

I'm not quite familiar with localization, sorry for asking vague questions.
Looking forward to your feedback.

OsamuSekino, 2020-06-30:
Thank you for your interest.

So you want to run eagleye using a rosbag that contains GNSS data such as NMEA and IMU data?

The reason we use RTKLIB is that we want to use the raw data because of the inherent filters in the receiver.
It is not recommended to use the navigation calculation results output from the receiver as it may show unintended behavior.

If you really want the eagleye to work with data from the receiver, you'll need to modify the eagleye.

1)
https://github.com/MapIV/eagleye/blob/master/eagleye_core/src/heading.cpp#L112
→For example, if it's NMEA data, put the azimuth of $GPVTG here. (rad)

2)
https://github.com/MapIV/eagleye/blob/master/eagleye_core/src/velocity_scale_factor.cpp#L84
→For example, if it's NMEA data, put the ground speed of $GPVTG here. (m/s)

3)
Publish the latitude and longitude altitude with the message type rtklib_msg::RtklibNav with the topic name /rtklib_nav.

After making these mods, eagleye should work if you enter the vehicle speed information(geometry_msgs::TwistStamped /can_twist) and IMU(sensor_msgs::Imu /imu/data_raw) data.

mcomic, 2020-07-14:
Thanks for your reply.

Following your advice, I did some modifies and tests during these days.
Now I'm able to send data to eagleye_core from my ROS script, but I got 2 more problems.

1. **Data was received, but no localization output**

[image]

The picture above is one step of the eagleye_core output after I sent my data.
What happening is:
->All the [status enable] is False.
->The position is not estimated.

I think the problem may cause by the data I use.
I've said that I have GNSS and IMU data before, but what I actually got are [Longitude, Latitude] and [vehicle velocity, acceleration of x, y, z] with the timestamp.

So about the code modify,
For https://github.com/MapIV/eagleye/blob/master/eagleye_core/src/heading.cpp#L112
-> I'm not sure what to put as azimuth here, therefore I didn't modify.
For https://github.com/MapIV/eagleye/blob/master/eagleye_core/src/velocity_scale_factor.cpp#L84
-> I put [vehicle velocity] in [ecef_vel.x] so the formula here will get [vehicle velocity] as the result.

And I've tried to lower the [estimated_velocity_threshold] and [estimated_number_min] in the eagleye_config.yaml file, still the same problem.

Could you spot what is the cause of no localization output?
(If more information is needed, please tell me.)

2. **Can not find output file.**

This happens when I test with sample data either. After 100s of the test, I can not find any file that contains output data in eagleye_core folder and my catkin_ws root folder. And seems there is no further information about this in the Getting started documentation.
Could you please tell me where can I find the output file?

OsamuSekino, 2020-07-14:
Answer for 1.

>  I think the problem may cause by the data I use.
>  I've said that I have GNSS and IMU data before, but what I actually got are [Longitude, Latitude] and [vehicle velocity, acceleration of x, y, z] with the timestamp.

Estimated results are not output because eagleye does not have the necessary data.
GNSS => Time of Weeks(Or a time stamp.),Longitude, Latitude,Altitude,Velocity vector (azimuth,ground speed)
IMU => angular velocity of z
Vechicle => Speed
At a minimum, this information is required for eagleye. If it's not enough, no estimation result will be output.
Sorry, but this data does not work with the current eagleye.

Answer for 2.
If you want the output of the kml file, you can use
`rosrun fix2kml fix2kml`
This command must be executed. The output is kml.

mcomic, 2020-07-14:
Thanks for your answer.
Pity to know the current data doesn't suit. I will check if there is a way to get the data that lacks.

closed 2020-07-15 by OsamuSekino