MapIV/eagleye #57: Could the code work only based on GNSS and IMU?

weisongwen, 2020-09-13:
Thanks for sharing such a good package. Could it work when the velocity information is not available from the CAN?

Best,
Weisong,

kitsukawa, 2020-09-14:
@weisongwen At this point, velocity information from the vehicle CAN is mandatory. CAN-less Eagleye is also under development, so please wait for future updates for a while.

weisongwen, 2020-09-14:
hi @kitsukawa , Thanks for your update! I will keep an eye on this continuously. :-)

Best,
Weisong,

closed 2020-09-14 by kitsukawa

weisongwen, 2021-05-11:
Hi @kitsukawa , does the current package still requires the vehicle CAN for fusion? Since the CAN data is not usually available.

Best,
Weisong,

weisongwen, 2021-05-11:
Dear @rsasaki0109  Thanks for your update :-)

rsasaki0109, 2022-09-07:
We developed eagleye to work with only gnss and imu.

https://github.com/MapIV/eagleye/pull/138
https://github.com/MapIV/eagleye/pull/156