MapIV/eagleye #61: abs or std::abs ?

ssssjiang, 2021-03-24:
Hello 
（Sorry, my English is pool...）
In distance.cpp line 38
used "abs"
function abs is from stdlib.h，it only supports "int"
function std::abs is from c++ "cmath", it's able to support "double" "float" "long double"

Is this your intention?
But when I run code by eagleye_sample.bag , distance ---> height and pitch were zero for a long time

I don't know if this is normal...

Aoki-Takanose, 2021-03-29:
Thank you for pointing this out.
As you said, the correct intention is to use std::abs.
We will fix it in the next update.

As for the height and pitch not being output, we are using RTK positioning solution for these calculations.
The "eagleye_sample.bag" does not contain the RTK positioning solution, so it is not calculated.

MapIV also provides a bag file that includes RTK positioning data, so please feel free to use it.
https://www.dropbox.com/sh/mhdib1m1oivotiu/AAD0UnANDsuIsKqcSKHt9WAJa?dl=0

ssssjiang, 2021-03-29:
Thank you for your reply！

closed 2021-03-29 by ssssjiang