MapIV/eagleye #82: "inputting RTK results from the F9P" need two separate C099-F9P boards?

ideafold, 2021-10-30:
"When inputting RTK results from the F9P into the Eagleye, **two F9Ps** are used as follows.
(1) A receiver that outputs NMEA (RTK results from the F9P internal engine)
(2) A receiver that measures RAW data through RTKLIB and outputs Doppler velocity."
as this said.

by the way, "A receiver that measures RAW data", the RAW data is the RTCM V3 raw data ?
or rtcm v3 data can be used to calc velocity?

Thanks

rsasaki0109, 2021-10-30:
The raw data here should be in any format that can be positioned by rtklib, but map4 uses ubx and spf files, and We have not tested it with rtcm.

```
inpstr1-format     =ubx        # (0:rtcm2,1:rtcm3,2:oem4,3:oem3,4:ubx,5:ss2,6:hemis,7:skytraq,8:gw10,9:javad,10:nvs,11:binex,12:rt17,13:sbf,14:cmr,15:tersus,18:sp3)
```
https://github.com/MapIV/RTKLIB/blob/rtklib_ros_bridge/app/rtkrcv/conf/rtklib_ros_bridge_meijo_rtk.conf#L14

rsasaki0109, 2021-10-30:
> "inputting RTK results from the F9P" need two separate C099-F9P boards?

Yes.

ideafold, 2021-11-03:
> > "inputting RTK results from the F9P" need two separate C099-F9P boards?
> 
> Yes.

Thank you for the kindly reply.
why need two separate C099-F9P boards?
one C099-F9P board can output both raw data and NMEA?

rsasaki0109, 2021-11-03:
This is because when we did that before, the output of f9p became abnormal.

ideafold, 2021-11-03:
thanks.
If it can use only one f9p, it will be great.
By the, MapIV/eagleye's RTK function can work well with ubox NEO-M8T? and still need two NEO-M8T?

rsasaki0109, 2021-11-03:
We have not tested it with m8t, but maybe eagleye will work with m8t.
I don't know if it can be done with one m8t, but if one m8t can output nmea and ubx, eagleye will work.

Also, m8t is a single-frequency receiver, so I don't recommend it because of its poor rtk performance.

sukhrajklair, 2022-02-03:
Are both gnss mounted on the vehicle? How do you get rtk corrections when both are moving?

rsasaki0109, 2022-03-10:
Both GNSS are mounted on the vehicle.
Correction information is also obtained from another GNSS reference station.
Three GNSS receivers are required.

rsasaki0109, 2022-07-06:
This issue is closed for now. If you have problems again, please reopen this issue.

closed 2022-07-06 by rsasaki0109