MapIV/eagleye #91: rtkrcv server start error

IvanGerov, 2021-11-29:
I am getting an error related to the inpstr1-path. When I run the node, the first thing that I get is rtkrcv server start error where it cannot find the receiver. 

Solutions that I've tried:

- renamed inpstr1-path to match ttyACM0 and ttyUSB0, still the same error
- tried the sample inpstr1-path by MapIV and only changed the baud rate, still the same error

I am assuming I have to input the serial number of the chip somewhere in the config file? At least that's what I understood from the instructions. Any suggestions are highly valued, thank you!

NikolayDonchevski, 2021-12-02:
I have the same error.

rsasaki0109, 2021-12-03:
You may not have access to the serial port.
Please use chmod or "sudo usermod -G dialout -a $USER" to get access to the serial port.

NikolayDonchevski, 2021-12-03:
Thank you very much for your suggestion. I tried that and I still get the same error:
"rtk server start error (str1 open error path=/serial/by-id/usb-u-blox_AG_-_www.u-blox.com_u-blox_GNSS_receiver-if00:115200:8:n:1:off)"

when I run the command: "sudo setserial /dev/ttyACM0"

I get the following message which I think is a problem:
"/dev/ttyACM0, UART: unknown, Port: 0x0000, IRQ: 0"

Is there any sort of drivers I need to install?
I am using U blox ZED-F9P on Ubuntu 18.04 with ROS melodic

rsasaki0109, 2022-03-10:
Sorry to say this now, but you may want to reboot your PC.

rsasaki0109, 2022-07-06:
This issue is closed for now. If you have problems again, please reopen this issue.

closed 2022-07-06 by rsasaki0109