MapIV/eagleye #92: Assistance regarding executing the package

dineshrboson, 2021-12-11:
Hi team thanks for the great package.
I am using ublox zed f9p module GPS.
I am getting navsat fix with ublox ros package
I am getting speed from wheel sensor via can message and then to ros.
I am getting IMU from zed camera.
I want to achieve localization using these with your package.
but when I tried I saw rtklib_nav is working fine but nmea launch files are not working properly.
instead of using rtklib nav and nmea can I launch kumar robotics ublox package where I will get /ublox/fix which consists gps coordinates.
eagleye:                                           #GNSS cycle 5Hz, IMU cycle 50Hz.

# Using Topic Name
 twist_topic: /can_twist
 imu_topic: /imu/data_raw
 rtklib_nav_topic: /rtklib_nav
 navsatfix_topic: /f9p/fix
 nmea_sentence_topic: /f9p/nmea_sentence

if its available what changes I had to make in this conf file

dineshrboson, 2021-12-13:
[image]

dineshrboson, 2021-12-13:
[image]

Module I am using

rsasaki0109, 2022-03-10:
Since /ublox/fix is also sensor_msgs/NavSatFix type, it can be used.
However, some functions such as pitch angle estimation are not available.

rsasaki0109, 2022-07-06:
This issue is closed for now. If you have problems again, please reopen this issue.

closed 2022-07-06 by rsasaki0109