MapIV/eagleye #129: ros2-foxy build error

cshang412, 2022-05-07:
Current branch **main-ros2**

~$ **colcon build --packages-up-to eagleye_rt --symlink-install**
Starting >>> geographic_msgs
Starting >>> rtklib_msgs
Starting >>> nmea_msgs
Finished <<< rtklib_msgs [0.59s]                                                                                                   
Starting >>> eagleye_msgs
Finished <<< nmea_msgs [0.95s]                                                                                                  
Finished <<< geographic_msgs [1.41s]                                                               
Starting >>> geodesy
Finished <<< geodesy [0.30s]                                                            
Starting >>> geographic_info
Finished <<< eagleye_msgs [1.22s]                                                             
Finished <<< geographic_info [0.19s]                  
Starting >>> eagleye_coordinate
Finished <<< eagleye_coordinate [0.27s]                  
Starting >>> eagleye_navigation
Finished <<< eagleye_navigation [0.43s]                     
Starting >>> eagleye_rt
[Processing: eagleye_rt]                             
[Processing: eagleye_rt]                                     
--- stderr: eagleye_rt                                         
In file included from /usr/include/c++/9/ext/alloc_traits.h:36,
                 from /usr/include/c++/9/bits/basic_string.h:40,
                 from /usr/include/c++/9/string:55,
                 from /usr/include/c++/9/bits/locale_classes.h:40,
                 from /usr/include/c++/9/bits/ios_base.h:41,
                 from /usr/include/c++/9/ios:42,
                 from /usr/include/c++/9/ostream:38,
                 from /usr/include/c++/9/iostream:39,
                 from /home/david/Documents/ros2/localization/gnss_imu/install/geodesy/include/geodesy/utm.h:43,
                 from /home/david/Documents/ros2/localization/gnss_imu/install/eagleye_coordinate/include/eagleye_coordinate/eagleye_coordinate.hpp:30,
                 from /home/david/Documents/ros2/localization/gnss_imu/src/eagleye-1.3.0-ros2/eagleye_rt/src/enable_additional_rolling_node.cpp:31:
/usr/include/c++/9/bits/alloc_traits.h: In instantiation of ‘struct std::allocator_traits<std::allocator<const std::shared_ptr<eagleye_msgs::msg::VelocityScaleFactor_<std::allocator<void> > >&> >’:
/opt/ros/foxy/include/rclcpp/message_memory_strategy.hpp:47:9:   required from ‘class rclcpp::message_memory_strategy::MessageMemoryStrategy<const std::shared_ptr<eagleye_msgs::msg::VelocityScaleFactor_<std::allocator<void> > >&, std::allocator<void> >’
/opt/ros/foxy/include/rclcpp/node_impl.hpp:91:1:   required by substitution of ‘template<class MessageT, class CallbackT, class AllocatorT, class CallbackMessageT, class SubscriptionT, class MessageMemoryStrategyT> std::shared_ptr<SubscriptionT> rclcpp::Node::create_subscription(const string&, const rclcpp::QoS&, CallbackT&&, const rclcpp::SubscriptionOptionsWithAllocator<AllocatorT>&, typename MessageMemoryStrategyT::SharedPtr) [with MessageT = eagleye_msgs::msg::VelocityScaleFactor_<std::allocator<void> >; CallbackT = void (&)(const std::shared_ptr<eagleye_msgs::msg::VelocityScaleFactor_<std::allocator<void> > >&); AllocatorT = std::allocator<void>; CallbackMessageT = const std::shared_ptr<eagleye_msgs::msg::VelocityScaleFactor_<std::allocator<void> > >&; SubscriptionT = rclcpp::Subscription<const std::shared_ptr<eagleye_msgs::msg::VelocityScaleFactor_<std::allocator<void> > >&, std::allocator<void>, rclcpp::message_memory_strategy::MessageMemoryStrategy<const std::shared_ptr<eagleye_msgs::msg::VelocityScaleFactor_<std::allocator<void> > >&, std::allocator<void> > >; MessageMemoryStrategyT = rclcpp::message_memory_strategy::MessageMemoryStrategy<const std::shared_ptr<eagleye_msgs::msg::VelocityScaleFactor_<std::allocator<void> > >&, std::allocator<void> >]’
/home/david/Documents/ros2/localization/gnss_imu/src/eagleye-1.3.0-ros2/eagleye_rt/src/enable_additional_rolling_node.cpp:135:142:   required from here
/usr/include/c++/9/bits/alloc_traits.h:399:13: error: forming pointer to reference type ‘const std::shared_ptr<eagleye_msgs::msg::VelocityScaleFactor_<std::allocator<void> > >&’
  399 |       using pointer = _Tp*;
      |             ^~~~~~~
/usr/include/c++/9/bits/alloc_traits.h:402:13: error: forming pointer to reference type ‘const std::shared_ptr<eagleye_msgs::msg::VelocityScaleFactor_<std::allocator<void> > >&’
  402 |       using const_pointer = const _Tp*;
      |             ^~~~~~~~~~~~~
In file included from /usr/include/c++/9/bits/locale_conv.h:41,
                 from /usr/include/c++/9/locale:43,
                 from /usr/include/c++/9/iomanip:43,
                 from /home/david/Documents/ros2/localization/gnss_imu/install/geodesy/include/geodesy/utm.h:44,
                 from /home/david/Documents/ros2/localization/gnss_imu/install/eagleye_coordinate/include/eagleye_coordinate/eagleye_coordinate.hpp:30,
                 from /home/david/Documents/ros2/localization/gnss_imu/src/eagleye-1.3.0-ros2/eagleye_rt/src/enable_additional_rolling_node.cpp:31:
/usr/include/c++/9/bits/unique_ptr.h: In instantiation of ‘struct std::default_delete<const std::shared_ptr<eagleye_msgs::msg::VelocityScaleFactor_<std::allocator<void> > >&>’:
/opt/ros/foxy/include/rclcpp/message_memory_strategy.hpp:119:18:   required from ‘class rclcpp::message_memory_strategy::MessageMemoryStrategy<const std::shared_ptr<eagleye_msgs::msg::VelocityScaleFactor_<std::allocator<void> > >&, std::allocator<void> >’
/opt/ros/foxy/include/rclcpp/node_impl.hpp:91:1:   required by substitution of ‘template<class MessageT, class CallbackT, class AllocatorT, class CallbackMessageT, class SubscriptionT, class MessageMemoryStrategyT> std::shared_ptr<SubscriptionT> rclcpp::Node::create_subscription(const string&, const rclcpp::QoS&, CallbackT&&, const rclcpp::SubscriptionOptionsWithAllocator<AllocatorT>&, typename MessageMemoryStrategyT::SharedPtr) [with MessageT = eagleye_msgs::msg::VelocityScaleFactor_<std::allocator<void> >; CallbackT = void (&)(const std::shared_ptr<eagleye_msgs::msg::VelocityScaleFactor_<std::allocator<void> > >&); AllocatorT = std::allocator<void>; CallbackMessageT = const std::shared_ptr<eagleye_msgs::msg::VelocityScaleFactor_<std::allocator<void> > >&; SubscriptionT = rclcpp::Subscription<const std::shared_ptr<eagleye_msgs::msg::VelocityScaleFactor_<std::allocator<void> > >&, std::allocator<void>, rclcpp::message_memory_strategy::MessageMemoryStrategy<const std::shared_ptr<eagleye_msgs::msg::VelocityScaleFactor_<std::allocator<void> > >&, std::allocator<void> > >; MessageMemoryStrategyT = rclcpp::message_memory_strategy::MessageMemoryStrategy<const std::shared_ptr<eagleye_msgs::msg::VelocityScaleFactor_<std::allocator<void> > >&, std::allocator<void> >]’
/home/david/Documents/ros2/localization/gnss_imu/src/eagleye-1.3.0-ros2/eagleye_rt/src/enable_additional_rolling_node.cpp:135:142:   required from here
/usr/include/c++/9/bits/unique_ptr.h:71:9: error: forming pointer to reference type ‘const std::shared_ptr<eagleye_msgs::msg::VelocityScaleFactor_<std::allocator<void> > >&’
   71 |         default_delete(const default_delete<_Up>&) noexcept { }

cshang412, 2022-05-07:
**update:**  i got it, the issue was caused by callback function locate in`[ enable_additional_rolling_node.cpp](https://github.com/MapIV/eagleye/blob/24774cf2530d009a9580043777d7db3283abf42a/eagleye_rt/src/enable_additional_rolling_node.cpp#L49)`

For example:
**change it from** 
`void velocity_scale_factor_callback(const eagleye_msgs::msg::VelocityScaleFactor::**ConstPtr &msg**)
{
  _velocity_scale_factor = *msg;
}`

**to**

`void velocity_scale_factor_callback(const eagleye_msgs::msg::VelocityScaleFactor::**ConstSharedPtr msg**)
{
  _velocity_scale_factor = *msg;
}`

rsasaki0109, 2022-05-07:
Thanks for the report. Correction.

rsasaki0109, 2022-05-08:
Fixed a bug and added CI with ros foxy. Thanks for the report. Please contact me if you have any more problems.
Now that the problem has been resolved, close this issue.

closed 2022-05-08 by rsasaki0109