MapIV/eagleye #151: Canless eagleye

laitathei, 2022-07-19:
Is eagleye_rt_canless.launch means that the package now are supporting only using GNSS and IMU data to estimate the position without out twist_topic?

rsasaki0109, 2022-07-19:
@laitathei 
Yes, it is.
Note that canless mode may be less accurate than normal mode in non-open sky environments.

laitathei, 2022-07-19:
Thank you. Also, I want to know that do you used any NTRIP Client service to provide RTK position or any other methods in your demo video? Because I got the same module Ublox ZED-F9P and I want to get the same result as your demo video GNSS green line.

rsasaki0109, 2022-07-19:
@laitathei 
The eagleye demo video does not use RTK, but this is a video with twist_topic.
In the case of canless, RTK must be used.

rsasaki0109, 2022-07-21:
The issue in the title has been resolved, so I'm closing this issue.
If you have problems using canless, please open a new issue.

closed 2022-07-21 by rsasaki0109