MapIV/eagleye #159: Bag not work for ros2

cshang412, 2022-09-11:
when using `ros2 bag play -s rosbag_v2 eagleye_sample.bag` play bag , the **rosbag_v2** only support default msgs that  included in ros1 and ros2 standard msg, the  **rtklib_msgs/RtklibNav** msg will be  ignored.

1.  is there a way to not using  **rtklib_msgs/RtklibNav** msg ,just using **sensor_msgs/NavSatFix**  to replace gnss source as input ?

rsasaki0109, 2022-09-11:
To play rosbag in ros1 with custom messages in ros2, please see below.
https://github.com/ros2/rosbag2_bag_v2

rtklib_msgs/RtklibNav contains GNSS Doppler velocity (velocity in ECEF coordinate system) and sensor_msgs/NavSatFix is not a replacement.

I would like to provide sample data from ROS2 sometime in the future.

cshang412, 2022-09-13:
thanks very much for your quick response, May i ask one more question, Since the  GNSS Dopper velocity is not common used in normal gnss device, i can only get GPRMC or GNRMC sentence which is presice position after differentiation positioning, and also i can get chassis can frame and imu frame , i want use GPRMC or GNRMC to replace the RTK input , in NMEA mode, what the difference between `/nmea_sentence_topic` and `gga_topic` ？ and both of them should have input at same time ? i that neccessary ?

rsasaki0109, 2022-09-13:
gga_topic has disappeared in the develop branch.
So, please disregard this matter. 
https://github.com/MapIV/eagleye/blob/develop-ros1/eagleye_rt/config/eagleye_config.yaml#L8-L12

If you do not use Doppler velocity, set use_gnss_mode from RTKLIB to NMEA and specify nmea_sentence_topic with gga and rmc.

By the way, ealgeye internally converts nmea_sentence_topic(nmea_msgs/Sentence) to nmea_msgs/Gpgga and nmea_msgs/Gprmc.

rsasaki0109, 2022-10-17:
I forgot to mention that the following package developed by map4 also provides ros1/ros2 conversion for rosbag.

https://gitlab.com/MapIV/rosbags

rsasaki0109, 2022-11-07:
Because it has not had recent activity, I am closing this issue for now. Please comment again if you have any further issues.

closed 2022-11-07 by rsasaki0109

rsasaki0109, 2023-04-18:
New sample data is here.
https://drive.google.com/drive/folders/1TEaVRn7F6t8PhEfZe59gE_C1B8jZUkJo

enescingoz, 2024-02-14:
> New sample data is here. https://drive.google.com/drive/folders/1TEaVRn7F6t8PhEfZe59gE_C1B8jZUkJo

Is there any pcd and osm data to test this dataset using Autoware?