MapIV/eagleye #228: How long does IMU only solution can stand

ajay1606, 2023-03-09:
Hello,

Thank you so much for very impressive work and haring source.

After going through, i just curious to know, how long does localization node stable using only imu odometry ? when there is no GPS and No NDT output.

Existing autoware Localization module needs position input either from GPS or NDT, failing of both (Ex: Inside long straight Tunnel, nieghter GPS, nor NDT works)  would result in immediate failure of entire localization module.

Alternatively, if IMU only can handle for at least shorter period of time would be more beneficial in order to perform vehicle emergency maneuvers. 

Regards,
Ajay

rsasaki0109, 2023-03-09:
Without GNSS, eagleye performs dead reckoning with vehicle speed and IMU and its performance is 50cm error at 100m.
Localizaiton with GNSS/IMU localization in areas without much GNSS is not recommended without LiDAR or Visual assistance.

ajay1606, 2023-03-09:
@rsasaki0109 Thank you for your quick response.

closed 2023-03-09 by rsasaki0109