MapIV/eagleye #279: Add heading information from driver

tiesus, 2023-06-08:
First of all thanks for the amazing package. 
If have a question regarding the heading source. In dual antenna setups you are fusing the input from both antennas seperately. 
I understand that the driver is intended to work with raw values through rtk lib instead of driver specific messages. 
For me it would be interesting to fuse heading from the driver directly instead of using multi antenna mode. Is this possible? 
My intention to use this package this way is to use params like antenna::hading offset.

rsasaki0109, 2023-06-08:
Instead of launching gnss_compass.launch, you can input a topic called gnss_compass_pose of the type geometry_msgs::msg::PoseStamped to the heading_node. There is no need to input a position into this gnss_compass_pose topic.

Comment out the following gnss_compass.launch.
https://github.com/MapIV/eagleye/blob/main-ros2/eagleye_rt/launch/eagleye_rt.launch.xml#L140-L145

Please subscribe to the driver posture below.
https://github.com/MapIV/eagleye/blob/main-ros2/eagleye_rt/src/heading_node.cpp#L243

tiesus, 2023-06-09:
This is exactly what I'm  looking for:) Thanks for your quick response.