MapIV/eagleye #281: ROS Packages uses wrong path for installation

leochien1110, 2023-06-16:
Hi,

I would like to confirm that the path of `gsigeo2011_ver2_1.asc`. I followed step but there is not such file in the path `eagleye/eagleye_util/llh_converter/data/gsigeo2011_ver2_1.asc`. It turns out the file is under `llh_converter/data/gsigeo2011_ver2_1.asc`. 

This issue also appears in `ros1-main` branch. I have update the step in my [fork](https://github.com/leochien1110/eagleye) but not sure if it is correct. Please let me know if I could create a PR.

Thank you for all these efforts :slightly_smiling_face:

rsasaki0109, 2023-06-16:
You are correct. Thank you for pointing out the error in the file path. Please go ahead and create a pull request

leochien1110, 2023-07-05:
@rsasaki0109  Just create a PR, please check.
Another issue is that this problem also happens in other branches. Though it is just a tiny fix, that will result in many PRs. Should I let your team do the rest?

rsasaki0109, 2023-07-05:
For now, the main branch is sufficient. We will do the rest.

rsasaki0109, 2023-07-28:
I have also fixed ROS1 and will close this issue.
https://github.com/MapIV/eagleye/commit/abf0647e6a36be78db8242ae14399616a3064630

closed 2023-07-28 by rsasaki0109