MapIV/eagleye #288: Position seems to be offsetted in Eagleye. What could be causing this?

sisaha9, 2023-07-11:
The yellow dot in the below video is the ground truth and the blue dot is the estimated location provided by Eagleye

https://github.com/MapIV/eagleye/assets/49205122/10b83259-6615-42a9-b26d-2ff77ff84c93

Here is the parameter file being used

```yaml
/**: #GNSS cycle 5Hz, IMU cycle 50Hz.
  ros__parameters:
    # Estimate mode
    use_gnss_mode: rtklib
    use_can_less_mode: false
    use_multi_antenna_mode: true

    # Topic
    twist:
      twist_type: 0 # TwistStamped : 0, TwistWithCovarianceStamped: 1
      twist_topic: /eagleye_input/can_twist
    imu_topic: /novatel_interface/imu
    gnss:
      velocity_source_type: 3 # rtklib_msgs/RtklibNav: 0, nmea_msgs/Sentence: 1, ublox_msgs/NavPVT: 2, geometry_msgs/TwistWithCovarianceStamped: 3
      velocity_source_topic: /novatel_interface/twist
      llh_source_type: 2 # rtklib_msgs/RtklibNav: 0, nmea_msgs/Sentence: 1, sensor_msgs/NavSatFix: 2
      llh_source_topic: /novatel_interface/fix
    sub_gnss:
      llh_source_type: 2 # nmea_msgs/Sentence: 1, sensor_msgs/NavSatFix: 2
      llh_source_topic:  /sensing/sub_gnss/ublox/nav_sat_fix

    # TF
    tf_gnss_frame:
      parent: "base_link"
      child: "novatel"

    # Origin of GNSS coordinates (ECEF to ENU)
    ecef_base_pos:
      x: 0.0
      y: 0.0
      z: 0.0
      use_ecef_base_position: false

    reverse_imu_wz: false

    # Navigation Parameters
    # Basic Navigation Functions
    common:
      imu_rate: 100
      gnss_rate: 20
      stop_judgment_threshold: 0.01
      slow_judgment_threshold: 0.278
      moving_judgment_threshold: 2.0

    velocity_scale_factor:
      estimated_minimum_interval: 20
      estimated_maximum_interval: 400
      gnss_receiving_threshold: 0.25
      save_velocity_scale_factor: false
      velocity_scale_factor_save_duration: 100.0
      th_velocity_scale_factor_percent: 10.0

    yaw_rate_offset_stop:
      estimated_interval: 3
      outlier_threshold: 0.002

    yaw_rate_offset:
      estimated_minimum_interval: 30
      gnss_receiving_threshold: 0.25
      outlier_threshold: 0.002
      1st:
        estimated_maximum_interval: 300
      2nd:
        estimated_maximum_interval: 500

    heading:
      estimated_minimum_interval: 10
      estimated_maximum_interval: 30
      gnss_receiving_threshold: 0.25
      outlier_threshold: 0.0524
      outlier_ratio_threshold: 0.5
      curve_judgment_threshold: 0.0873
      init_STD: 0.0035 #[rad] (= 0.2 [deg])

    heading_interpolate:
      sync_search_period: 2
      proc_noise: 0.0005 #[rad] (= 0.03 [deg])

    slip_angle:
      manual_coefficient: 0.0

    slip_coefficient:
      estimated_minimum_interval: 2
      estimated_maximum_interval: 100
      curve_judgment_threshold: 0.017453
      lever_arm: 0.0

    rolling:
      filter_process_noise: 0.01
      filter_observation_noise: 1

    trajectory:
      curve_judgment_threshold: 0.017453
      timer_update_rate: 10
      deadlock_threshold: 1
      sensor_noise_velocity: 0.05
      sensor_scale_noise_velocity: 0.02
      sensor_noise_yaw_rate: 0.01
      sensor_bias_noise_yaw_rate: 0.1

    smoothing:
      moving_average_time: 3
      moving_ratio_threshold: 0.1

    height:
      estimated_minimum_interval: 200
      estimated_maximum_interval: 2000
      update_distance: 0.1
      gnss_receiving_threshold: 0.1
      outlier_threshold: 0.3
      outlier_ratio_threshold: 0.5
      moving_average_time: 1

    position:
      estimated_interval: 300
      update_distance: 0.1
      outlier_threshold: 3.0
      gnss_receiving_threshold: 0.25
      outlier_ratio_threshold: 0.5

    position_interpolate:
      sync_search_period: 2

    # Optional Navigation Functions
    angular_velocity_offset_stop:
      estimated_interval: 4
      outlier_threshold: 0.002

    rtk_dead_reckoning:
      rtk_fix_STD: 0.3 #[m]
      proc_noise: 0.05 #[m]

    rtk_heading:
      update_distance: 0.3
      estimated_minimum_interval: 10
      estimated_maximum_interval: 30
      gnss_receiving_threshold: 0.25
      outlier_threshold: 0.0524
      outlier_ratio_threshold: 0.5
      curve_judgment_threshold: 0.0873

    enable_additional_rolling:
      update_distance: 0.3
      moving_average_time: 1
      sync_judgment_threshold: 0.01
      sync_search_period: 1

    velocity_estimator:
      gga_downsample_time: 0.5
      stop_judgment_velocity_threshold: 0.2
      stop_judgment_interval: 1
      variance_threshold: 0.000025
      pitch_rate_offset:
        estimated_interval: 8
      pitching:
        estimated_interval: 5
        outlier_threshold: 0.0174
        gnss_receiving_threshold: 0.2
        outlier_ratio_threshold: 0.5
      acceleration_offset:
        estimated_minimum_interval: 30
        estimated_maximum_interval: 500
        filter_process_noise: 0.01
        filter_observation_noise: 1
      doppler_fusion:
        estimated_interval: 4
        gnss_receiving_threshold: 0.2
        outlier_ratio_threshold: 0.5
        outlier_threshold: 0.1
```

sisaha9, 2023-07-11:
Not sure if this should be in a separate issue but it doesn't look like the RPY get's updated either as seen in the pose on the top left panel that comes in the screen a couple of times

rsasaki0109, 2023-07-11:
Is it possible to have us provide rosbag?

sisaha9, 2023-07-11:
Hi @rsasaki0109 . Here is a link: https://drive.google.com/drive/folders/1nmsM7Fz6pjkn_Nj_x4qW8Ee0N8TxVvRi?usp=sharing . It is right now shared with the public email on your Github profile

rsasaki0109, 2023-07-12:
Thank you. I'm busy during the weekdays this week so I'll look at it in my free time.