MapIV/eagleye #291: How can I use Eagleye in MGRS coordinates

meliketanrikulu, 2023-07-13:
I want to see my position in MGRS coordinates. I changed this launch file configuration --> https://github.com/MapIV/eagleye/blob/94228d3b646b82d24e7d5565f819c82ed4113abc/eagleye_util/fix2pose/launch/fix2pose.launch.xml#L12C1-L12C1 (I set tf_num = 2). But its not working . I tested with example rosbag .

Here is my output: 
[image]
 You can see here '/eagleye/pose' topics output on terminal and RVIZ .

How can I get my position in MGRS coordinates.

rsasaki0109, 2023-07-13:
Is eagleye/fix outputting the correct latitude and longitude?
I would like to see what the eagleye terminal output looks like.

rsasaki0109, 2023-07-18:
Later, each branch will reflect this, but the following modifications will work.

bafore
```
  if(_enu_absolute_pos.status.enabled_status)
```
after
```
  if(_enu_absolute_pos.status.enabled_status && (_eagleye_fix.latitude != 0.0 || _eagleye_fix.longitude != 0.0))
```
https://github.com/MapIV/eagleye/commit/741032a60ea0060353defcb1d5fb61f4f93d119c

meliketanrikulu, 2023-07-28:
Now its not give result in this --> (eagleye/fix) topic. But still its still not working in MGRS coordinates.

rsasaki0109, 2023-07-29:
Is the fix2pose node dead? Could you check the fix2pose log file in the .ros directory to identify what the issue might be? Here is the link: https://github.com/MapIV/eagleye/tree/main-ros2/eagleye_util/fix2pose