MapIV/eagleye #292: Eagleye is not woring when use_can_less_mode is false

meliketanrikulu, 2023-07-14:
I tested with default test data and default parameter. Just I set use_can_less_mode parameter as false. It not give an output in /eagleye/fix topic. Here is the eagleye terminal output. Can I use eagleye without vehicle twist msg. How can I fix this issue

[image]

rsasaki0109, 2023-07-15:
Sorry, the CanLess can only operate with the following data. We will eventually turn this data into a sample rosbag.
https://drive.google.com/drive/folders/1TEaVRn7F6t8PhEfZe59gE_C1B8jZUkJo

meliketanrikulu, 2023-07-19:
> Sorry, the CanLess can only operate with the following data. We will eventually turn this data into a sample rosbag. https://drive.google.com/drive/folders/1TEaVRn7F6t8PhEfZe59gE_C1B8jZUkJo

I cant understand why its depends on test data. What is the difference between this data and sample data in the README.md

rsasaki0109, 2023-07-19:
sample rosbag is a rosbag with a special navsatfix that says No RTK-FIX when status is -1 and RTK FIX when status is 0. It does not support the current eagleye canless mode.
If you convert the navsatfix to nmea_msgs/Gpgga, you can run it in sample rosbag.

meliketanrikulu, 2023-07-28:
Okey thank you for information. @rsasaki0109  Maybe you can add this on README.md file.

closed 2023-07-28 by meliketanrikulu