MapIV/eagleye #293: raw gnss data and eagleye output look mirrored to each other

meliketanrikulu, 2023-07-18:
I visualize raw gnss data and eagleye's output (navsatfix msgs ) as local cartesian UTM. For your sample rosbag data output is as we wait . Sample data output is here:
[image]
The red one is raw gnss data output and blue one is eagleye output (/eagleye/fix) pose.

But I have interesting problem when i tested with my data. The poses looks mirrored. Here is the results:
[image]

My calibration is correct . I am using default parameters .

rsasaki0109, 2023-07-18:
Does setting [`reverse_imu_wz`](https://github.com/MapIV/eagleye/blob/main-ros2/eagleye_rt/config/eagleye_config.yaml#L34) to true not solve the problem?

issue
https://github.com/MapIV/eagleye/issues/289

meliketanrikulu, 2023-07-18:
> Does setting [`reverse_imu_wz`](https://github.com/MapIV/eagleye/blob/main-ros2/eagleye_rt/config/eagleye_config.yaml#L34) to true not solve the problem?
> 
> issue #289

Yes I tested when [reverse_imu_wz](https://github.com/MapIV/eagleye/blob/main-ros2/eagleye_rt/config/eagleye_config.yaml#L34)  is false. It gives same output

rsasaki0109, 2023-07-18:
Is the coordinate system of the input GNSS velocity enu?

rsasaki0109, 2023-07-19:
> Yes I tested when [reverse_imu_wz](https://github.com/MapIV/eagleye/blob/main-ros2/eagleye_rt/config/eagleye_config.yaml#L34) is false. It gives same output

I think the trajectories will be different depending on whether reverse_imu_wz is set to true or false. Your comment implies that both trajectories deviate from the output of GNSS data, is that correct?

rsasaki0109, 2023-07-19:
I would like to be provided with rosbag if the above issues seem to be ok.

meliketanrikulu, 2023-07-19:
> I think the trajectories will be different depending on whether reverse_imu_wz is set to true or false. Your comment implies that both trajectories deviate from the output of GNSS data, is that [...]

I tested both of the [reverse_imu_wz](https://github.com/MapIV/eagleye/blob/main-ros2/eagleye_rt/config/eagleye_config.yaml#L34) settings. I set true and false . It give same output

meliketanrikulu, 2023-07-19:
> I would like to be provided with rosbag if the above issues seem to be ok.
Okey Thank you. I will provide a rosbag .

meliketanrikulu, 2023-07-19:
> The following functions can be used to convert from enu-based velocities to ecef-based velocities.

Thank you. Our velocities are not in the ECEF coordinates. It might be my problems reason. I will share bag file after I convert my velocities to ECEF and test with that.

meliketanrikulu, 2023-07-28:
Hello @rsasaki0109 . Its solved when we give ECEF velocity. Thank you for your support .

closed 2023-07-28 by meliketanrikulu