MapIV/eagleye #301: cannot play sample data ROS 1 with ROS2 foxy environment

wafeez, 2023-07-27:
Hi, 

I have problem to run the sample data in ROS2 (foxy) environment. The error are as below:

**Files:             eagleye_sample.bag
Bag size:          3.5 GiB
Storage id:        rosbag_v2
Duration:          1673.60s
Start:             Jan 27 2020 12:12:52.310 (1580098372.310)
End:               Jan 27 2020 12:40:45.370 (1580100045.370)
Messages:          252597
Topic information: Topic: /can_twist | Type: geometry_msgs/msg/TwistStamped | Count: 135508 | Serialization Format: rosbag_v2
                   Topic: /imu/data_raw | Type: sensor_msgs/msg/Imu | Count: 83653 | Serialization Format: rosbag_v2
                   Topic: /fix | Type: sensor_msgs/msg/NavSatFix | Count: 8360 | Serialization Format: rosbag_v2**

Please note that for both ROS1(noetic) and ROS2(foxy), the msg missing has been installed.But still cant play the bag. Please help to advice.

rsasaki0109, 2023-07-28:
It seems that there are no custom messages in the checked message, so how are you playing the rosbag? I think it would be good to convert rosbag in ros1 to ros2 using rosbags.
https://pypi.org/project/rosbags/

rsasaki0109, 2023-07-28:
We have prepared new samples, and they include data from both ros1 and ros2.
https://drive.google.com/drive/folders/1TEaVRn7F6t8PhEfZe59gE_C1B8jZUkJo

rsasaki0109, 2025-03-29:
This is our company's Google Drive, so I can only respond on weekdays.

joeyjyyang, 2025-04-11:
> We have prepared new samples, and they include data from both ros1 and ros2. https://drive.google.com/drive/folders/1TEaVRn7F6t8PhEfZe59gE_C1B8jZUkJo

The link is broken for me too. Can you please make the ROS 2 sample bag(s) accessible? Thanks!