MapIV/eagleye #302: can_less mode with custom dataset

thatnn, 2023-08-01:
Thank you for your great work

I can't apply can less mode other rosbag with /nmea_sentence and /imu/data like below.

[image]

here are  rosbag and my eagleye_config.yaml.

error for imu frame_id :"/base_imu", I republish frame_id: "imu" to /imu/data2

rosbag: https://drive.utbm.fr/s/6NcE2GSqNdGyELg

eagleye_config.yaml.

# Estimate mode
use_gnss_mode: NMEA
use_can_less_mode: True

# Topic
twist:
  twist_type: 0 # TwistStamped : 0, TwistWithCovarianceStamped: 1
  twist_topic: /can_twist
imu_topic: /imu/data2
gnss:
  velocity_source_type: 1 # rtklib_msgs/RtklibNav: 0, nmea_msgs/Sentence: 1, ublox_msgs/NavPVT: 2, geometry_msgs/TwistWithCovarianceStamped: 3
  velocity_source_topic:  /nmea_sentence
  llh_source_type: 1 # rtklib_msgs/RtklibNav: 0, nmea_msgs/Sentence: 1, sensor_msgs/NavSatFix: 2
  llh_source_topic: /nmea_sentence
sub_gnss:
  llh_source_type: 1 # nmea_msgs/Sentence: 1, sensor_msgs/NavSatFix: 2
  llh_source_topic: /nmea_sentence
# TF
tf_gnss_frame:
  parent: "base_link"
  child: "gnss"

reverse_imu_wz: false
reverse_imu_ax: false

# Origin of GNSS coordinates (ECEF to ENU)
ecef_base_pos:
  x : 0.0
  y : 0.0
  z : 0.0
  use_ecef_base_position : false

# Eagleye Navigation Parameters
# Basic Navigation Functions
common:
  imu_rate: 50
  gnss_rate: 5
  stop_judgement_threshold: 0.01
  slow_judgement_threshold: 0.278
  moving_judgement_threshold: 2.78

velocity_scale_factor:
  estimated_minimum_interval: 20
  estimated_maximum_interval: 400
  gnss_receiving_threshold: 0.25
  velocity_scale_factor_save_str: /config/velocity_scale_factor.txt
  save_velocity_scale_factor: false
  velocity_scale_factor_save_duration: 100.0
  th_velocity_scale_factor_percent: 10.0

yaw_rate_offset_stop:
  estimated_interval: 4
  outlier_threshold: 0.002

yaw_rate_offset:
  estimated_minimum_interval: 30
  gnss_receiving_threshold: 0.25
  outlier_threshold: 0.002
  1st:
    estimated_maximum_interval: 300
  2nd:
    estimated_maximum_interval: 500

heading:
  estimated_minimum_interval: 10
  estimated_maximum_interval: 30
  gnss_receiving_threshold: 0.25
  outlier_threshold: 0.0524
  outlier_ratio_threshold: 0.5
  curve_judgement_threshold: 0.0873
  init_STD: 0.0035 #[rad] (= 0.2 [deg])

heading_interpolate:
  sync_search_period: 2
  proc_noise: 0.0005 #[rad] (= 0.03 [deg])

slip_angle:
  manual_coefficient: 0.0

slip_coefficient:
  estimated_minimum_interval: 2
  estimated_maximum_interval: 100
  curve_judgement_threshold: 0.017453
  lever_arm: 0.0

rolling:
  filter_process_noise: 0.01
  filter_observation_noise: 1

trajectory:
  curve_judgement_threshold: 0.017453
  timer_updata_rate: 10
  deadlock_threshold: 1
  sensor_noise_velocity: 0.05
  sensor_scale_noise_velocity: 0.02
  sensor_noise_yaw_rate: 0.01
  sensor_bias_noise_yaw_rate: 0.1

smoothing:
  moving_average_time: 3
  moving_ratio_threshold: 0.1

height:
  estimated_minimum_interval: 200
  estimated_maximum_interval: 2000
  update_distance: 0.1
  gnss_receiving_threshold: 0.1
  outlier_threshold: 0.3
  outlier_ratio_threshold: 0.5
  moving_average_time: 1

position:
  estimated_interval: 300
  update_distance: 0.1
  outlier_threshold: 3.0
  gnss_receiving_threshold: 0.25
  outlier_ratio_threshold: 0.5
  gnss_error_covariance: 0.5

position_interpolate:
  sync_search_period: 2
  proc_noise: 0.05 #[m]

monitor:
  print_status: true
  log_output_status: false
  update_rate: 10.0
  th_gnss_deadrock_time: 10.0
  use_compare_yaw_rate: false
  comparison_twist_topic: /calculated_twist
  th_diff_rad_per_sec: 0.17453
  th_num_continuous_abnormal_yaw_rate: 25
  th_dr_distance: 50.0

# Optional Navigation Functions
angular_velocity_offset_stop:
  estimated_interval: 4
  outlier_threshold: 0.002

rtk_dead_reckoning:
  rtk_fix_STD: 0.3 #[m]
  proc_noise: 0.05 #[m]

rtk_heading:
  update_distance: 0.3
  estimated_minimum_interval: 10
  estimated_maximum_interval: 30
  gnss_receiving_threshold: 0.25
  outlier_threshold: 0.0524
  outlier_ratio_threshold: 0.5
  curve_judgement_threshold: 0.0873

enable_additional_rolling:
  update_distance: 0.3
  moving_average_time: 1
  sync_judgement_threshold: 0.01
  sync_search_period: 1

velocity_estimator:
  gga_downsample_time: 0.5
  stop_judgement_velocity_threshold: 0.2
  stop_judgement_interval: 1
  variance_threshold: 0.000025
  pitchrate_offset:
    estimated_interval: 8
  pitching:
    estimated_interval: 5
    outlier_threshold: 0.0174
    gnss_receiving_threshold: 0.2
    outlier_ratio_threshold: 0.5
  acceleration_offset:
    estimated_minimum_interval: 30
    estimated_maximum_interval: 500
    filter_process_noise: 0.01
    filter_observation_noise: 1
  doppler_fusion:
    estimated_interval: 4
    gnss_receiving_threshold: 0.2
    outlier_ratio_threshold: 0.5
    outlier_threshold: 0.1

rsasaki0109, 2023-08-01:
Eagleye has static initialization → moving initialization, and we look at the following topics:

/eagleye/yaw_rate_offset_stop topic
When the "enabled_status" of this topic becomes true, the static initialization (stopping for a few seconds, regardless of indoors or outdoors) is completed.
/eagleye/heading_xx
Dynamic initialization requires straight-line travel in an environment where GNSS speed is available (it's okay if a curve is included in the middle of the travel).
For straight-line travel, there is a lower speed limit called "moving_judgment_threshold," and by default, it is set to 10 km/h. For low-speed vehicles, it might be acceptable to reduce it to 3-5 km/h.
[Here's the link to the configuration](https://github.com/MapIV/eagleye/blob/main-ros2/eagleye_rt/config/eagleye_config.yaml#L43).

thatnn, 2023-08-02:
> Eagleye has static initialization → moving initialization, and we look at the following topics:
> 
> * /eagleye/yaw_rate_offset_stop topic
>   When the "enabled_status" of this topic becomes true, the static initialization (stopping for a few seconds, regardless of indoors or outdoors) is completed.
> * /eagleye/heading_xx
>   Dynamic initialization requires straight-line travel in an environment where GNSS speed is available (it's okay if a curve is included in the middle of the travel).
>   For straight-line travel, there is a lower speed limit called "moving_judgment_threshold," and by default, it is set to 10 km/h. For low-speed vehicles, it might be acceptable to reduce it to 3-5 km/h.
>   [Here's the link to the configuration](https://github.com/MapIV/eagleye/blob/main-ros2/eagleye_rt/config/eagleye_config.yaml#L43).
> 
> I will look at the rosbag when I have time.

Thank you for your reply

I changed moving_judgment_threshold: 2.78 to 0.556 or 0.834

but nothing better 

I'm still waiting for your reply.

Thank you

rsasaki0109, 2023-08-02:
Visualise the time series and trajectory of the angular velocity wx, speed and gnss rtk status of the imu.