MapIV/eagleye #317: Bad performance on example bag in ros2

yenlunliao, 2023-12-07:
Hello,
I am working on the eagleye on ros2 branch.
I plot the result by Foxglove, but found the localization is not stable.

The result was showed as follow, 
Purple: localization from gnss
Blue: localization from eagleye
[image]

The localization is matched perfect initially but having the jump somewhere and misdirection after few seconds.
Any Suggestion or idea to fix it, 
Thanks.

Environment: Ubuntu 20.04/foxy, Ubuntu 20.04/humble
Config: eagleye_config.yaml in main-ros2 branch
Launch: eagleye_rt.launch.xml in main-ros2 branch
Bag: ros2 in https://drive.google.com/drive/folders/1TEaVRn7F6t8PhEfZe59gE_C1B8jZUkJo

rsasaki0109, 2023-12-08:
Sorry, the sample IMU has a pitch of 180 degrees, so please correct the tf. I will also modify the code later.
https://github.com/MapIV/eagleye/blob/main-ros2/eagleye_util/tf/launch/tf.launch.xml#L11

yenlunliao, 2023-12-11:
I change the transform to 
```
<push-ros-namespace namespace="imu"/>
        <node name="static_transform_publisher" pkg="tf2_ros" exec="static_transform_publisher" args="0 0 0 0 3.14159265 0 base_link imu">
        </node>
```

and it performs well, thanks.

closed 2023-12-11 by yenlunliao