MapIV/eagleye #324: How to use yaw_rate_offset_parameter.estimated_minimum_interval and estimated_maximum_interval

itootas, 2024-03-27:
Hi,
I'm not sure how to use estimated_minimum_interval and estimated_maximum_interval in yaw rate offset parameter.

According to https://github.com/MapIV/eagleye/tree/main-ros1/eagleye_rt/config, these are set as the follow for default.

yaw_rate_offset:
  estimated_minimum_interval: 30
  1st:
    estimated_maximum_interval: 300
  2nd:
    estimated_maximum_interval: 500

I understand these parameters mean the time of yaw rate offset estimation and also offset is decided after two times of least squares estimation, but I don't when they are applied. Could you tell me more details?

In the case of default for 1st estimation, I guess as bellow behavior. Is this correct?
 t: driving time
i) t<30s ; not estimate
ii) t=30s ; estimate with gnss velocity and yaw rate data for last 30s
iii) 30s<t<=300s ; estimate with data for last t[s]
iv) t>300s ; estimate with data for last 300s
  
BR,

rsasaki0109, 2024-03-27:
> In the case of default for 1st estimation, I guess as bellow behavior. Is this correct?

That's correct.

For more details, it would be good to look at the paper titled "Eagleye: A Lane-Level Localization Using Low-Cost GNSS/IMU" from the IV2021 workshop.

itootas, 2024-03-27:
Thank you for your quick support. I'd like to read the paper.

By the way, I think 300s is a little long for estimation. Is it typical to estimate while driving?

rsasaki0109, 2024-03-27:
The yaw rate offset is also estimated beforehand using the average at rest. 
300 seconds is not the minimum but the maximum, so I don't think it's that long. The main target is situations where it's not possible to estimate the yaw rate offset using the average at rest for a long time, such as on highways.

itootas, 2024-03-27:
>  The main target is situations where it's not possible to estimate the yaw rate offset using the average at rest for a long time, such as on highways.
I understand. So, you mean the offset is also estimated at vehicle stop like ZUPT (zero velocity update)?
Priority of the offset at rest is higher than at run and yaw rate offset is rarely estimated by maximum time on local road?

itootas, 2024-03-27:
Let me one more question,
this estimation method is popular? I mean, is it common knowledge to estimate for more than 100s while long driving such as highways?

rsasaki0109, 2024-03-27:
> I understand. So, you mean the offset is also estimated at vehicle stop like ZUPT (zero velocity update)?

Yes, that's right.

> Priority of the offset at rest is higher than at run and yaw rate offset is rarely estimated by maximum time on local road?

As for the priority of the offset, it seems that the error changes slightly between being at rest and in motion. During driving, the yaw rate offset estimated while in motion is prioritized.

rsasaki0109, 2024-03-27:
> Let me one more question,
this estimation method is popular? I mean, is it common knowledge to estimate for more than 100s while long driving such as highways?

I'm sorry, I don't quite understand... Please contact the author of the paper.

itootas, 2024-03-28:
Understood, my question has been closed. Thanks a lot.

closed 2024-03-28 by itootas