MapIV/eagleye #339: can_less mode and single point positioning

nisshi3, 2025-02-09:
Thank you for your great work.

I would like to use Eagleye in can_less mode with stand-alone positioning.
I am using simpleRTK2B (u-blox ZED-F9P ) and IMU (TAG300N-100).

(1)Does Eagleye_rt need both nmea_sentence and rtklib_msgs inputs? 
    It is difficult to input nmea_sentence and rtklib_msgs to Eagleye_rt at the same time because UART2 is not available on the F9P board we are using.

(2) I am inputting GNSS and IMU Topic with nmea_sentence specified, but /velocity is not published. What could be the cause of this problem?

(3) I am using the ROSBAG (eagleye_sample_20200127.bag) to check if it localization in CANLES mode, but it does not.
 What could be the cause of this problem?
(I have specified rtklib_msgs as the GNSS to be used.)

rsasaki0109, 2025-02-12:
(1) If you set use_gnss_mode to NMEA, it should work with only the nmea_sentence as the GNSS input.
https://github.com/MapIV/eagleye_private/blob/main-ros1/eagleye_rt/config/eagleye_config.yaml#L3
(2) Setting use_gnss_mode to NMEA should make it work.
(3) In eagleye_sample_20200127.bag, the CAN-less mode does not work.

nisshi3, 2025-02-13:
Thank you for your response.
(1), (2) I will verify the operation by setting use_gnss_mode to NMEA.
If it does not work, I will extract the ROSBAG data.
(3) Why does the CAN-less mode not work?
Are there any limitations, such as requiring GNSS: Ublox F9P with RTK for the CAN-less mode?

rsasaki0109, 2025-02-13:
The following indicates that canless requires RTK positioning.
https://map-iv.slack.com/archives/C072JDQHPBN/p1736838992589089?thread_ts=1736810754.458759&cid=C072JDQHPBN