MapIV/eagleye #340: Failed to lookup transform: "base_link" passed to lookupTransform argument target_frame does not exist.

aditya-yamaha, 2025-02-11:
I tried running the sample files from dropbox (eagleye_sample_20200127.bag) on the ros1 branch of the repo. On running the  roslaunch eagleye_rt eagleye_rt.launch command, it throws an error:

[image]

How to resolve this?

This happens on either no changes to yaml as well as changing the topics in eagleye_config.yaml

[image]

rsasaki0109, 2025-02-12:
You should be good to go if you change "gnss" to "gps."
https://github.com/MapIV/eagleye/blob/main-ros1/eagleye_util/tf/config/sensors_tf.yaml#L22

aditya-yamaha, 2025-03-19:
@rsasaki0109 Thanks for the help, the issue lies in a grammatical mistake itself in sensor_tf.yaml source code https://github.com/MapIV/eagleye/blob/main-ros1/eagleye_util/tf/config/sensors_tf.yaml, where it has got the word parent misspelled as patent. On correcting it along with the right frame_id, building the pkg again, seems to solve the issue.

closed 2025-03-19 by aditya-yamaha