MapIV/eagleye #341: NMEA Sentences Compatibilty

sthakur-ymsli, 2025-02-12:
My ROS bag contains NMEA sentences, including GPGSV, GNRMC, and GNGGA. I reviewed the `gnss_converter_node` package and found that it only processes GPGGA and GPRMC sentences. I'm unsure whether it will accept input from my NMEA topic to execute its internal logic. Could you please provide guidance on how we can use our NMEA topic as input to the package and what changes should be made.

[image]

[image]

rsasaki0109, 2025-02-12:
If you set use_gnss_mode to NMEA instead of RTKLIB, it should work.

https://github.com/MapIV/eagleye/blob/main-ros1/eagleye_rt/config/eagleye_config.yaml#L3

sthakur-ymsli, 2025-02-12:
We are using Autoware main branch. Did all the required changes in the config. file of the package. We are using NMEA instead of RTK, all the input topics is publishing data and the package is subscribing them. But still on the terminal, we are not getting "status enable: True" for the nodes in the pacakge architecture, except yaw_rate offset stop. Not able to figure out the issue. Could you please help and guide. FYR: We played our rosbag for an hour still no change in enable status.

[image]

rsasaki0109, 2025-02-12:
Could you provide the coming YAML?

sthakur-ymsli, 2025-02-13:
[image]
[image]
[image]
[image]
[image]

rsasaki0109, 2025-02-13:
I want to check the original differences, so I need the YAML file as it is. At a glance, it seems fine.
By the way, what’s the typical speed of the vehicle? Self-localization doesn’t work unless it’s running at more than 10 kph.

sthakur-ymsli, 2025-02-13:
A little unclear on what you refer by the Original YAML file but here's the link for that:
https://github.com/MapIV/eagleye/blob/autoware-main/eagleye_rt/config/eagleye_config.yaml

and our YAML file is in the pictures above.

Coming to the vehicle's speed, we estimate it to be around 6-14 kph. 

My question to you is that, do we need the vehicle to operate above 10kph at all times or is it just an initialisation thing? If yes, what could be the workaround to achieve package functionality?

rsasaki0109, 2025-02-13:
I would like to check where the YAML file has been modified.

It is necessary to always drive at a speed of over 10 km/h. Of course, it is possible to stop temporarily to some extent.
In that case, please reduce the moving_judgment_threshold from 2.78 [m/s] to half.
https://github.com/MapIV/eagleye/blob/autoware-main/eagleye_rt/config/eagleye_config.yaml#L39

sthakur-ymsli, 2025-02-13:
Our IMU is giving the data at 10 Hz, GPS is also around 10 Hz, NMEA is around 2 Hz. Twist is giving at 30 Hz. I tried changing the rate in YAML file to 30 Hz for both but the orientation algorithm won't converge. Your guidance would be helpful, kindly suggest the alternatives.

[image]

rsasaki0109, 2025-02-13:
Could you set the gnss_rate in the YAML file to 2?

sthakur-ymsli, 2025-02-14:
Thanks for the help, we will try with the suggested configuration.