MapIV/eagleye #343: Unable to get lat,long and altitude data in eagleye rt

aditya-yamaha, 2025-03-19:
Hello,

All my values of lat,long and latitude always point to 0 in eagleye status in input.

[image]

I'm using the [MSC RAD4R dataset.
](https://mscrad4r.github.io/loop/overall/) (LOOP_D0.bag)

Here is the info on the different topics available on the bag:

[image]

Also, here is my config file:

[image]

I Tried both NMEA and RTKLIB in use_gnss_mode. Also, tried inter-changing llh_source topics in gnss and sub_gnns but still nothing.

Can you please point out to where the issue might be.

rsasaki0109, 2025-03-19:
use_gnss_mode can be set to RTKLIB mode without any issues.
However, ublox fix velocity is of type TwistStamped, while Eagleye requires the input to be of type TwistWithCovarianceStamped.

aditya-yamaha, 2025-03-19:
@rsasaki0109 Thank you for the reply. 

1) ublox fix velocity is already is of the type TwistWithCovarianceStamped not TwistStamped.

[image]

2) Even on setting use_gnss_mode to RTKLIB, I still face the same issue.

No values other than 0 in lat,long and altitude.

rsasaki0109, 2025-03-19:
The issue is that the following is not working properly. This is the output from --RTKLIB--, and you need to check if there are any issues around that area. Since TOW is present, I suspect there might be something wrong with the input topic.
https://github.com/MapIV/eagleye/blob/autoware-main/eagleye_util/gnss_converter/src/gnss_converter_node.cpp#L116-L143

aditya-yamaha, 2025-03-20:
@rsasaki0109 

I think I found the issue. On echoing the topic **/eagleye/navsat/rtklib_nav**, I got:

[image]

Which clearly shows that status is not populated properly even though all input topics from the bag are correct.

On further investigation, I found out that in the gnss_converter_node.cpp might have some bugs.
Here in this snippet from main-ros1 branch, there seems to be no clear referencing of the r.status, which causes fields to be left zero.

If we were to add another line (  r.status = *nav_msg_ptr; ) in the function shown below,

[image]

We can see now the topic is populated correctly and eagle status also shows the data being populated.

[image]

[image]

If my finding is correct, I'll raise a PR to fix the issue.

rsasaki0109, 2025-03-20:
Sounds good! Please create a PR.
It would be good to add an if statement similar to ROS 2, like in the following example:
https://github.com/MapIV/eagleye/blob/autoware-main/eagleye_util/gnss_converter/src/gnss_converter_node.cpp#L121-L124