MapIV/eagleye #348: NMEA sentence compatibility

taiyangkaodot, 2025-04-02:
I have a question.
Currently, my NMEA sentences include gpfpd and gtimu, where the former corresponds to speed and latitude/longitude, while the latter corresponds to IMU data.

[image]

Now I want to use eagleye for positioning and need to modify some parameters. Currently, our autonomous driving test vehicle speed is around 10 km/h.

[image]
The first question is, if the first option is selected for Veloity_source_type, our NMEA sentences have two formats, will it automatically recognize the speed?
The second question is, how should we roughly adjust the other parameters? Our car is a GAC Trumpchi GE3.

taiyangkaodot, 2025-04-02:
Also, what does sub_gnss mean? We only have one set of integrated navigation hardware.
Thank you in advance, everyone.

rsasaki0109, 2025-04-02:
The sub GNSS is an optional feature using a second antenna and is not mandatory. The NMEA sentences must include both GGA and RMC, and you need to change the gnss XXX_type setting and set use_gnss_mode to nmea.

taiyangkaodot, 2025-04-02:
Thank you for your prompt reply, but I'm not quite clear—what do the GGA and RMC sentences mean? Currently, in this GNSS ROS2 driver, there are two sentences, the first starting with $GTIMU.

[image] As shown in the image, it depicts the IMU. So, for the first GNSS configuration related to velocity, do I also need to write a conversion logic to read the raw data from the GNSS driver's sentences, convert it into a geometry/twist type topic data with velocity and covariance, and then have Eagleye subscribe to it?

taiyangkaodot, 2025-04-02:
thanks a lot

taiyangkaodot, 2025-04-02:
Thank you, there should be one last question. I see that my driver can enable dual-antenna mode, so which part of the configuration file still needs to be modified? Additionally, my test vehicle is quite slow, around 0-10 km/h. Should the threshold be lowered, otherwise the function package might behave abnormally?